Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #6: How to export all the labels result?

austingg, 2018-07-31:
How to export the labels ?  sorry for I didn't find where to export all the label json. 
And the `Show Json output` button failed to show the json 😢

austingg, 2018-07-31:
The server says:
```
W20180731-15:24:10.580(8)? (STDERR) (node:560) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'socName' of undefined
W20180731-15:24:10.581(8)? (STDERR)     at generateJson (server/api.js:32:43)
W20180731-15:24:10.582(8)? (STDERR)     at /home/kafei/.meteor/packages/promise/.0.10.2.oel1k0.5yqbh++os+web.browser+web.cordova/npm/node_modules/meteor-promise/fiber_pool.js:43:40
```

dmandrioli, 2018-07-31:
I can reproduce this problem clicking the JSON button right after loading an image without adding any polygons, this is a bug I will create an issue for that. Did you get this error message doing the same or after adding some polygons?

austingg, 2018-07-31:
@dmandrioli  click after I finish labelling

dmandrioli, 2018-07-31:
So this is a bug, I will close the issue because there is no way to investigate more. Feel free to re-open 
 with explanation if you know how to reproduce. Thanks for your feedback.

closed 2018-07-31 by dmandrioli

austingg, 2018-08-01:
For reproduce the problem,
on Chrome., when Click `Json Output` after add polygon, the browser get a new link
`http://10.10.106.2:8888/api/json/person%2Fperson_0041.jpg` 

and the sever log
```
W20180801-09:55:01.618(8)? (STDERR) (node:4682) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'socName' of undefined
W20180801-09:55:01.619(8)? (STDERR)     at generateJson (server/api.js:32:43)
W20180801-09:55:01.619(8)? (STDERR)     at /home/kafei/.meteor/packages/promise/.0.10.2.oel1k0.5yqbh++os+web.browser+web.cordova/npm/node_modules/meteor-promise/fiber_pool.js:43:40
```
either use the default classes set or custom classes set.

austingg, 2018-08-01:
after I  remove the chinease characters in the request url, the json result has successful shown.