Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #10: Corrupted download image

HemaZ, 2018-08-11:
hello, 
in the beginning, I really want to thank you for this great tool. it saved me a lot of time and effort 

i got a problem when i press the download button after I segmented an image.

it is shown in this screenshot 

[image]

dmandrioli, 2018-08-11:
Hey, are you using Firefox? If yes, your issue can be related to #2, you may want to try with Chrome/Chromium to see if the problem occurs.

HemaZ, 2018-08-21:
@dmandrioli I tried chrome and the same happens. except the .json file output is okay but the output image is corrupted

HemaZ, 2018-08-21:
@dmandrioli  i tried several images and the same is happening. 
I'm running the editor using meteor on Ubuntu 18.04

closed 2018-08-28 by dmandrioli