Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #11: How can i define my own class name in the docker enviroment

EthonDeep, 2018-08-20:
hi Damien Mandrioli:
               i want to use this tools to building my own pcl dataset , on VM(ubuntu16.04),and windows 10 install ,all  fail,but the docker can run . i wanna change the calss name in label pcl ,how can do that!

thanks for your help!

dmandrioli, 2018-08-21:
Hi, using the Docker image you should be able to change classes using a settings.json like this
https://github.com/jshimko/meteor-launchpad/issues/15#issuecomment-346393110

EthonDeep, 2018-08-21:
my script:
           export METEOR_SETTINGS=$(cat /home/tom/semantic-segmentation-editor/settings.json)

sudo docker run -e METEOR_SETTINGS="$METEOR_SETTINGS" -p 2004:3000 -v /home/tom/Downloads/images:/mnt/images -v /home/tom/Downloads/pointcloud:/mnt/pcd hitachiail/semantic-segmentation-editor:latest

[-] External MONGO_URL not found. Starting local MongoDB...

Delaying startup for 5 seconds...

[-] External MONGO_URL not found. Starting local MongoDB...

Delaying startup for 5 seconds...
=> Starting app on port 3000...
Semantic Segmentation Editor
Images (JPG, PNG, PCD) folder: /mnt/images
3D Segmentation Data folder: /mnt/pcd
1 sets of object classes

it doesn't work ,the label nothing change,why?

EthonDeep, 2018-08-21:
@dmandrioli 

my script:
export METEOR_SETTINGS=$(cat /home/tom/semantic-segmentation-editor/settings.json)

sudo docker run -e METEOR_SETTINGS="$METEOR_SETTINGS" -p 2004:3000 -v /home/tom/Downloads/images:/mnt/images -v /home/tom/Downloads/pointcloud:/mnt/pcd hitachiail/semantic-segmentation-editor:latest

[-] External MONGO_URL not found. Starting local MongoDB...

Delaying startup for 5 seconds...

[-] External MONGO_URL not found. Starting local MongoDB...

Delaying startup for 5 seconds...
=> Starting app on port 3000...
Semantic Segmentation Editor
Images (JPG, PNG, PCD) folder: /mnt/images
3D Segmentation Data folder: /mnt/pcd
1 sets of object classes

it doesn't work ,the label nothing change,why?

dmandrioli, 2018-08-23:
It works as expected using 
<code>
docker run -p 80:3000 -e METEOR_SETTINGS="$(cat /path/to/settings.json)" hitachiail/semantic-segmentation-editor:latest
</code>

Using this command line, paths and classes are all defined in settings.json, no need to specify folders with -v

closed 2018-08-23 by dmandrioli