Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #17: load result json file

FYGitHub1009, 2018-09-01:
Thank you for good tool.
I would like to load auto-annotated json files.
Is it possible?

brappazzo, 2019-02-27:
Has anyone worked on this?

dmandrioli, 2019-02-28:
Hello @brappazzo, nothing new on this feature, feel free to provide a PR.
Thanks.

brappazzo, 2019-05-10:
I am trying to work on this, any suggestions on the best approach?

dmandrioli, 2019-05-10:
First draw some polygons then look at the content of the mongo database, the goal is to inject data with the same format.
Then you will need a server-side service to inject the JSON into mongo, you can take inspiration from server/files.js or server/api.js. The transfered data could be a regular file or directly the JSON content.

I hope this helps, I will be happy to review your pull request if you plan to submit one.