Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #19: no pointcloud visualization

JSRGFJZ, 2018-09-20:
Hi, all I've install npm with version 1.6, and it's ok, after command 
 "meteor npm install
meteor --settings settings.json",
and I sucesfully change the setting.jason, and push some pcd file into the folder, after I log into the web browser, why no pcd visualization.? this is strange.
and it appears in the terminal like:
W20180920-17:23:57.041(8)? (STDERR) Error: ENOENT: no such file or directory, stat '/home/jean/semantic-segmentation-editor/mnt/image2/table_scene_lms400.pcd.labels'
W20180920-17:23:57.137(8)? (STDERR) Error: ENOENT: no such file or directory, stat '/home/jean/semantic-segmentation-editor/mnt/image2/table_scene_lms400.pcd.objects'

dmandrioli, 2018-09-20:
Hi,

If I well understand you can see your PCD files in the browser view but the pointcloud is not visible in the 3D editor.

Do you have any message on the browser console? These error messages on the server console are "normal" when you open a new file.

You can try to put this [PCD file](https://github.com/dmandrioli/sse-extra/blob/master/pointcloud.pcd) in your folder and try to open it. If it works your problem is related to the format of your PCD file. 

Another remark, your folders in settings.json should not be into SSE project folder to avoid annoying refresh of the app on each file write operation.

JSRGFJZ, 2018-09-20:
@dmandrioli
 Hi,
first, it's my pcd format problem, but why is that, I've tried many datasets, only your are ok.
Second, I see that in the output folder, we have file "*.pcd.labels" & " *.pcd.objects", how can I later read them?
thank you very much

closed 2018-09-21 by dmandrioli