Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #20: Could you share your Dockerfile?

brucechou1983, 2018-09-27:
Thanks for sharing this great tool. I actually run into some problems when running the docker image. Can't find the Dockerfile in the repo and on dockerhub. Could you share it? Thanks!

dmandrioli, 2018-09-27:
Hi, the Docker image is based on https://github.com/jshimko/meteor-launchpad
As you can see it is a one-line Docker file

closed 2018-09-27 by dmandrioli

brucechou1983, 2018-09-27:
Thanks for sharing the base image. It seems to me that building the image takes a long time (>30min on my MBP). Did you rebuild it every time you made modifications?

dmandrioli, 2018-09-27:
Unfortunately you're right, it's very long to build and yes you have to rebuild the image if you modify the app source code.

brucechou1983, 2018-09-28:
@dmandrioli Thanks. I've successfully built and modified the app.