Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #24: All labels are downloaded when going back from editor to navigator

brucechou1983, 2018-10-30:
Hi, thanks for sharing this great tool. Currently the only way to navigate to next/previous image is going back to the browsing mode and select the image. I think it's fine because it only costs a small fraction of time comparing to finish a segmentation label. However, I found it become much slower when I have more than a thousand labels in the MongoDB. Then I checked the requests:

[image]

and also the preview of the response
[image]

It downloads every single record in the system every time when going back to the file browsing mode. So the whole system will become slower and slower over the time. I'm trying to figure out how to fix this issue, but I'm not familiar with some part of the tech stack used in this project. Would be grateful if someone could help.

dmandrioli, 2018-10-30:
Hey, thank you for your feedback. I don't reproduce, can you take a screenshot of some expanded items of this 5MB array? Also note that the docker image is not up to date and there is no plan to update it so you may want to use the current git version.

brucechou1983, 2018-10-31:
@dmandrioli thanks for the support. These items are "ALL" labels store is my collection.
[image]

I use `jshimko/meteor-launchpad:latest` (a one line dockerfile) to launch this app with current git version.

dmandrioli, 2018-11-01:
This is a weakness of the app that need some work. Thanks for reporting.

brucechou1983, 2018-11-02:
@dmandrioli Thanks for confirming the issue. Is there anything I can help?

dmandrioli, 2018-11-07:
Sure, pull requests are welcome :)

brucechou1983, 2018-11-07:
Since I don't use the all annotated feature, I simply forced the annotated variables to empty lists.

dmandrioli, 2018-11-08:
Great! It looks good, I will review and apply your merge request it in the next couple of days. 
Thank you for your contribution.

brucechou1983, 2018-11-08:
You're welcome. Thanks for sharing this great tool.

dmandrioli, 2018-11-10:
Merged! Thanks Bruce.

closed 2018-11-10 by dmandrioli