Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #26: Should only load information about one image when entering editor mode

brucechou1983, 2018-11-14:
Hi, I faced another performance issue. When label number approaching ten thousand I found the speed of entering editing mode becomes much slower again. Then I check the network and see this response.

[image]

I grep the keyword "removed" and found nothing meaningful. Could you give me a hint about how to fix this? Thanks.

brucechou1983, 2018-11-15:
No. I'm using this commit.

> 44979b38

I've check the changlog of v1.1.2. Seems like there are only some Meteor dependency changes. I would like to know where these responses come from so I could debug and fix thems.

dmandrioli, 2018-11-15:
As a matter of fact, these logs are [DDP](https://github.com/meteor/meteor/blob/master/packages/ddp/DDP.md) messages but I don't know why you are seeing them. And I don't have time to label 10000 images so it will be difficult to reproduce :)
Look. in the direction of subscriptions and publications like you did before.

brucechou1983, 2018-11-15:
@dmandrioli Thanks. I will look into DDP. BTW, I don't manually create 10000 labels, I use a model to 
label these images so I only have to fix some of them.

dmandrioli, 2020-04-17:
Fixed in 44979b38ce119ab6a75cf18b6df412da4eae1988

closed 2020-04-17 by dmandrioli