Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #40: How to link My Images added in "semantic-segmentation-editor-1.1.2\private\samples""  its not working

RajasekharKadasani, 2019-06-11:


RajasekharKadasani, 2019-06-11:
how can i load image form server

dmandrioli, 2019-06-11:
Hi, I don't get your question, please elaborate.

RajasekharKadasani, 2019-06-11:
sorry,
I did setup App and trying to use ,How can i use my images either from local/server to load in this application instead of default images

dmandrioli, 2019-06-11:
Edit settings.json. By default, images are served from HOME/sse-images, not private/samples
https://github.com/Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor#configuration-file-settingsjson

RajasekharKadasani, 2019-06-11:
actually i am new to this Meteor app,
i followed steps and did setup but after build i didnt have any folder "sse-images"

dmandrioli, 2019-06-11:
What do you see on the console when you start the app?

RajasekharKadasani, 2019-06-11:
thank you i got that folder now

closed 2019-06-11 by RajasekharKadasani

Sajjadmanal, 2019-06-13:
@RajasekharKadasani Where did you find it? I am unable to find.