Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #46: Unable to remove the annotation in Point Cloud tool.

Sajjadmanal, 2019-06-13:
Hi Team,
This is really an amazing tool. Good job.
I am facing one issue. If I select the points for a class (say Car) these are selected in red color. I can remove a particular point which does not belongs to this  Car class by using (-) button provided in the tool. All works perfect till here.
Now, when I finally add these points to the car class, these points color changes to blue. Now, If I want to remove any point (which was marked mistakenly) from this class, I cannot do that.
Could you please help with this?

dmandrioli, 2019-06-13:
Thanks Sajjad! Just to let you know, the team is very small, just me :)
Look to be a bug, thanks for reporting.

Sajjadmanal, 2019-06-13:
@dmandrioli Kudos to your amazing work...!! This is an amazing app.

May I get your contact details? I have a few questions. (I will edit this line later.)

dmandrioli, 2019-06-13:
Just sent you an email.

dmandrioli, 2021-01-04:
Not a bug. The solution is to assign the void class to mistakenly marked points.

closed 2021-01-04 by dmandrioli