Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #47: Could you add a way to see the point's origional color ?

foolybiggun, 2019-06-19:
I'm trying to annotate an xyzrgb pcd file (some plants) via this tool.But I found that it is difficult to distinguish where the background and where the leaves is.If the point has its origional color,it would be more easier.Maybe a botten to switch the view between colored view and labled view would help?

lelouedec, 2019-07-10:
Good morning, 
I have started to add the color to the visualization, I added it to the PCD loader but trying to find how to make the visualization to display it. Any hint or idea so I can do it? 
To load the color into the geometry object I did :
 ```
 color.push(parseFloat(line[offset.r]));
  color.push(parseFloat(line[offset.g]));
  color.push(parseFloat(line[offset.b]));
```

dmandrioli, 2019-07-11:
Hello Justin,

You may want to look at the display() method here:
https://github.com/Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor/blob/master/imports/editor/3d/SseEditor3d.jsx#L1882

It's a good example of how to fill the color array.

Best,
Damien

lelouedec, 2019-07-11:
I indeed found this file and function and modified it to get the colors passed through the loader. However it seems that if the data is not annotated it is by default set to gray. I gave up for now on this lead.

dmandrioli, 2019-08-01:
Replaced by #51

closed 2019-08-01 by dmandrioli