Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #51: output PCD doesn't included intensity

PerfectWayne, 2019-08-01:
Hi, 
Thank you for providing such a good tool.
When I output labeled pcd file from a unlabeled pcb including intensity, the output labeled data doesn't include intensity and only XYZ. Is it possible to keep XYZI when output pcd file?
I don't know how to do it, could you please help with this?

dmandrioli, 2019-08-01:
Hi, thanks for your kind words.
This would need additional code, there is already a request for the same kind of feature (RGB PCD).
Let's create a generic feature request.

MacBlub, 2019-08-12:
I currently implemented support for Intensity values with a nice slider. :) (and actually support for other fields, but the implementation is not generic so I wouldnt push these) However, I only tested for Intensity-values between 0-10000. 

Here you can switch between the Segmentation-mode and Intensity-mode. I think, with a bit of Reverse-Engineering, someone can also add Support for RGB clouds.

I can do a MergeRequest for this feature, if requested. It was kinda hacked together, so no gurantee that it works well.

dmandrioli, 2019-08-12:
Hi, yes please submit a pull request, even if it's not ready to be merged as-is. We can't see anything for now :) I may use your code as a starting point to provide the generic feature.
Thanks a lot!

dmandrioli, 2019-08-13:
Thanks, I will give it a try in the next couple of days. Could you also provide a pcd sample file to test it?
@PerfectWayne What do you think, does it fit your need?

PerfectWayne, 2019-08-14:
Yes, It fits my need. The Intensity-value range is enough for me. I am grateful for your assistance. The PCD sample you can download here. 
[9272999P7.pcd.zip](https://github.com/Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor/files/3500576/9272999P7.pcd.zip)

MacBlub, 2019-08-14:
Glad to hear that. 

I can't share my pcd files, so I would like to prefer testing with PerfectWayne's file. I tried the file and in principle it works. However, the range is quite big, making the slider cubersome to use. To adapt the value to your prefered Range: 

Change in File: imports/editor/3d/SseCameraToolbar.jsx
Line: 124-125
124                             min={0}
125                             max={30000}

To your prefered range values. Probably to 0 - 500
124                             min={0}
125                             max={500}

I think a field to insert the Minimum- and Maximum-Range is quite useful here, because the intensity-field varies from laser to laser quite a bit...

dmandrioli, 2020-02-13:
Available in v1.4.0

closed 2020-02-13 by dmandrioli