Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #52: how to change the code for get the information of color?

ChenXie93, 2019-08-07:
Thanks for providing such a great tool for point cloud data edit, but i want to konw is there any possible to change some code that i can get the information of color after processing?

dmandrioli, 2019-08-07:
Thanks, you can get the classes of points using the [API](https://github.com/Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor#api-endpoints). The mapping between classes and colors is defined in settings.json, so you can post-process your data to get colors instead of classes. Hope this helps.

ChenXie93, 2019-08-07:
@dmandrioli ， thanks for you reply. but i want to get my own color information not the color definition in settings.json. when point colud data is imported by the editor, is has the information of RGB, but the output data through  the editor can't be saved with the RGB. So i wonder if there any setting that could help me to save the output data with the same RGB which was input at first.

dmandrioli, 2019-08-07:
I see, it's not possible for now, there is a already a feature request registered for this (#51)
Thanks for your feedback.

closed 2019-08-07 by dmandrioli