Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #54: Load the json

haleqiu, 2019-08-29:
The situation is I get some semi-finished 2d segmentation images with json files. How can I load these image into my app in order to keep on labeling these images? 
The json files are also generated using this app.

dmandrioli, 2019-08-29:
Hi, nothing ready to use for this use-case, you will have to directly modify the content of the mongo database (using a script or the tool of your choice). I'm wondering why you get this kind of situation, if the segmentation json comes from the app, why are you not able to just continue the labeling where your stopped?

haleqiu, 2019-08-29:
@dmandrioli  In my circumstance, multiple people are working on labeling in different server. You know, someone leaves and someone stays; sometimes the server is not available anymore. 
Now I need to transfer the data from one server to another and continue labeling based on the prior work.

dmandrioli, 2019-08-29:
@haleqiu  Thanks for clarification.

closed 2019-08-29 by dmandrioli