Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #59: Disable download

amiralipour, 2019-09-26:
Is there any way to disable download bottoms in the upper right corner of UI?

dmandrioli, 2019-09-26:
For PCD editor, remove the div block starting at line 81 in https://github.com/Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor/blob/master/imports/editor/3d/SseToolbar3d.jsx

For bitmap editor, remove the div block starting at line 58 in https://github.com/Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor/blob/master/imports/editor/2d/SseToolbar2d.jsx

amiralipour, 2019-09-27:
I am using docker image, can not locate the files inside the image.

dmandrioli, 2019-09-27:
If you run it like [this](https://github.com/Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor#running-the-app-using-docker), images and pcd files should be in /mnt/images.
Please note that the docker image is quite outdated.

amiralipour, 2019-09-27:
Is it possible to put the Dockerfile and docker-compose files so that we can modify and rebuild a docker image.

dmandrioli, 2019-09-29:
The existing docker image has been made using  
https://github.com/jshimko/meteor-launchpad

closed 2019-09-29 by dmandrioli

amiralipour, 2019-10-01:
> The existing docker image has been made using
> https://github.com/jshimko/meteor-launchpad

> The existing docker image has been made using
> https://github.com/jshimko/meteor-launchpad

In my case it re-install some meteor packages every time trying to run "meteor npm start" , how can i replace it with a enterypoint.sh file?