Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #63: Several questions...

aviadper, 2019-11-25:
Hi :-)

1. I'm using a docker. where can I see the configuration json file?
2. Is that possible to annotate the lidar points while seeing the image?
3. I cannot save the labeling output file. It says that the image is corrupted.

Thanks you!

dmandrioli, 2019-11-30:
Hi, alignment between bitmap and lidar data is not implemented, no way to annotate them together.
The provided docker image is based on v1.0 which is outdated.
Investigating your docker issues would need to have a new image based on the last version of the tool.

Thanks

aviadper, 2019-12-01:
Thank you. I used this image:
docker pull hitachiail/semantic-segmentation-editor

Where can I find the new version?

dmandrioli, 2019-12-01:
As of today there is no up-to-date docker image embedding the last version of the tool (1.1.4)

aviadper, 2019-12-01:
Thank you.
Does the new version have alignment between bitmap and lidar data

dmandrioli, 2019-12-01:
It would be a nice feature, but alignment is not implemented.

closed 2019-12-07 by dmandrioli