Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #64: save pcd on server instead of downloading to client

Mi-Scz, 2019-11-27:
Hello!

I also wanted to thank you for your work and this awesome tool! It really enhances the labeling procedure! I wanted to do some adjustments such that all the point clouds can remain on the server side and client does not need to download the labeled point cloud.

I could already adjust the SseToolbar3d.jsx such that on the top right corner -PCD Output- is only one "download" button left (instead of two buttons for download file and text pcd). This was the easy part.

Now I wanted to adjust the code such that the labeled point cloud can be saved on the server. I am really new to meteor and react, so I would really appreciate your help or hints with this problem. If my understanding is right I need to change the following:

1. change Line 83 to 86 in SseToolbar3d.jsx to
```
<div className="hflex">
    {this.renderCommand("downloadFileCommand")}
</div>
```

this will lead to have only one button left. Clicking the Download file Button in the editor causes to call the `downloadFile()` in line 527 in SseEditor3d.jsx

2. Changing this line 527 in SseEditor3d.jsx from `window.open("/api/pcdfile" + this.props.imageUrl, "_blank");` to a command such that the pcd file in `/api/pcdfile" + this.props.imageUrl` will be saved on the server file system (e.g. to ~/pcd/).

Unfortunately I do not have a clue which command I can use there. I would really appreciate if some one could help me here.

Thanks a lot!

dmandrioli, 2019-11-30:
Hi, thank you for your kind feedback! 
Everything is already saved on the server, pointcloud annotations as compressed binary files. There is no ready to use command to save annotations as text file, you will have to write some server side code, this could be done by modifying [this file](https://github.com/Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor/blob/master/server/SseDataWorkerServer.js).
Hope this helps.

Mi-Scz, 2019-12-02:
Thank you for your answer. I already checked this file. Unfortunately I do not see the command where the files are saved on the servers file system. I can just see how the files are compressed and uncompressed. Can you give me a hint, how I can got those files on my servers file system?

Mi-Scz, 2019-12-19:
Anyone a suggestion?

dmandrioli, 2019-12-19:
In `objectToBytes` function from `SseDataWorkerServer.js`, Dumping the `obj` argument to a file should do the trick.

closed 2020-02-11 by dmandrioli