Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #68: How to parse label and object?

Darknesszlx, 2019-12-10:
/api/pcdtext/[PATH_TO_FILE]: (3D only) Get the labeling of a pcd file using 2 addditional columns: label and object

How to parse these 2 addditional columns: label and object?

dmandrioli, 2019-12-10:
Are you talking about loading a pre-labeled PCD file in the tool?

Darknesszlx, 2019-12-10:
No, after labeling a pcd file, it will generate pcd_file_name.pcd.labels and pcd_file_name.pcd.objects, I think these two files saved marked info, but I don't know how to parse them in python or other programming language. Did I misunderstand these two files?
[image]

dmandrioli, 2019-12-10:
pcd_file_name.pcd.labels and pcd_file_name.pcd.objects are binary compressed internal files, they are not supposed to be used. When you call the API you get a consolidated text PCD file with two additional columns.

Darknesszlx, 2019-12-10:
OK, thanks for your replay!

closed 2019-12-10 by dmandrioli