Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #71: Bug: Disappearing Pointcloud right after loading

MaXL130, 2019-12-19:
Hi everyone,

I encountered a (sadly not realiably reproducable) bug when opening pointclouds with the editor. Right after loading, the points do appear, but as soon as I either enter the viewport or move the mouse inside of it, the visualized points just disappear.

I am still able to select the points and do get the selection outlines. But the points stay invisible.

Has anyone encoutered this problem, too? I'm currently investigating but could not find the source yet.

MaXL130, 2019-12-20:
Sadly I am not allowed to share the data. I did some digging in the code. I found that when opening a .pcd file, the call filling the graphic buffers is run multiple times. For the first few times, everything works out fine. But later on, there suddenly appears to be a mismatch between the number of xyz coordinates and color values. This causes an exception and the buffer is never displayed.

For my purpose, as a very quick n dirty fix, it was sufficient to remove the callbacks of the loading call [here](https://github.com/Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor/blob/34411db1455ee0666a0927e860375e76d9f59f5b/imports/editor/3d/SseEditor3d.jsx#L1998).

I could not find out why there was broken data to begin with, as the loading does work fine for the first few "rounds".

dmandrioli, 2019-12-20:
Thanks for sharing. So I guess your purpose is just to display the pointcloud because if you remove these callbacks, you also remove the loading of labeling data.

MaXL130, 2019-12-20:
Not really, I am labeling the pointclouds, loading and saving working just as intended. However I am working with a slightly modified version right now. So the fact that it is working for might be due to that. Still, the disappearing also happended to me with the unmodified version of this tool.

So my "fix" might not work out in general, but the bug is still there...

dmandrioli, 2020-01-09:
Hi, I did many changes in the pointcloud editor. 
Maybe this solves your issue?

closed 2020-04-16 by dmandrioli