Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #80: Unable to read the pcd data.

TianchaoHuo, 2020-03-23:
Firstly thanks for sharing a nice tool. while using this editor for my own pcd file,  I can not get any points on the screen. But I can see the point cloud in the pcl_viewer tools instead. 
[image]

As you can see, the points can be detected by the editor however it can not be shown on the screen.
[image]

I assume that the problem is from the data type : ascii or binary. Do I need to slightly modify or configure  the editor files? 

Could you help me to solve this problem ? Looking forward to your reply, thanks in advance.

dmandrioli, 2020-03-23:
Hi, yes you may have to change your input file, supported fields are described [here](https://github.com/Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor#pcd-support).
Can you share the PCD file that fails to display?

TianchaoHuo, 2020-03-23:
> Hi, yes you may have to change your input file, supported fields are described [here](https://github.com/Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor#pcd-support).
> Can you share the PCD file that fails to display?

Thanks for your quick reply！ the PCD file can be download here:
https://drive.google.com/open?id=1qGiIK6G_jQ1GvOs8B2Qhn4JQVJdl32Uq

I think I this PCD file has binary format.

CecilHarvey, 2020-03-24:
Your PCD file has NaN points, which is currently not supported.
Could you try removing all NaN points from your PCD file?

> Thanks for your quick reply！ the PCD file can be download here:
> https://drive.google.com/open?id=1qGiIK6G_jQ1GvOs8B2Qhn4JQVJdl32Uq
> 
> I think I this PCD file has binary format.

TianchaoHuo, 2020-03-25:
> Your PCD file has NaN points, which is currently not supported.
> Could you try removing all NaN points from your PCD file?
> 
> > Thanks for your quick reply！ the PCD file can be download here:
> > https://drive.google.com/open?id=1qGiIK6G_jQ1GvOs8B2Qhn4JQVJdl32Uq
> > I think I this PCD file has binary format.

closed 2020-03-25 by TianchaoHuo