Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #90: Can't load rgb pointcloud right

AliceBobC, 2020-06-15:
Thank you for brilliant work.

I have a problem with loading colored pointcloud, it didn't show any color information in the editor.
My pcd file is like this, is it a format issue or something else?

# .PCD v.7 - Point Cloud Data file format
VERSION .7
FIELDS x y z rgb
SIZE 4 4 4 4
TYPE F F F F
COUNT 1 1 1 1
WIDTH 20991
HEIGHT 1
VIEWPOINT 0 0 0 1 0 0 0
POINTS 20991
DATA ascii
0 -0.04876 1.219 1.648931e-38
0.002329524 -0.04892 1.223 1.887956e-38
0.004659048 -0.04892 1.223 1.860623e-38
0.006965715 -0.04876 1.219 1.833323e-38
0.00928762 -0.04876 1.219 1.851654e-38
0.01160952 -0.04876 1.219 1.869984e-38
-0.002321905 -0.0464381 1.219 1.35449e-38
0 -0.0464381 1.219 1.842727e-38
0.002329524 -0.04659048 1.223 2.340182e-38
........

dmandrioli, 2020-06-15:
Thanks! The rgb field has to be an integer, not a float.

AliceBobC, 2020-06-16:
> Thanks! The rgb field has to be an integer, not a float.

Really excited to get your reply! 

Could you help provide a sample pcd file ? I have tried several types of rgb values ​​and they all failed.
I got stuck here for days.

closed 2020-06-16 by dmandrioli

dmandrioli, 2020-06-16:
It was bug, sorry. You should now be able to visualize colors for text PCD files like this one:
```
VERSION .7
FIELDS x y z rgb
SIZE 4 4 4 4
TYPE F F F F
COUNT 1 1 1 1
WIDTH 9
HEIGHT 1
VIEWPOINT 0 0 0 1 0 0 0
POINTS 9
DATA ascii
0 -0.04876 1.219 1.648931e-38
0.002329524 -0.04892 1.223 0xFF0000
0.004659048 -0.04892 1.223 0x00FF00
0.006965715 -0.04876 1.219 0xFFFFFF
0.00928762 -0.04876 1.219 0x00FF00
0.01160952 -0.04876 1.219 0x0000FF
-0.002321905 -0.0464381 1.219 0x00FFFF
0 -0.0464381 1.219 1.842727e-38 0x00FF00
0.002329524 -0.04659048 1.223 0xFF8800
```
Note that you have to click the ligh bulb button (bottom right) to display colors.

AliceBobC, 2020-06-17:
It works now ！ Thank you

commit e5cdd329 (2020-06-16): Fix #90 RGB is only supported on binary PCD