Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #91: Mouse right button in pointcloud doesn't work.

wuyushuwys, 2020-06-17:
Mouse right button functionality in pointcloud doesn't work, can anyone help me with that?

dmandrioli, 2020-06-17:
So you can't select points using mouse right button drag&drop? Does something happen when you do this?

wuyushuwys, 2020-06-17:
No, I can't select points using mouse right button. When I dragged my mouse with pressing right mouse button, it would act like the reset the view. I built it from source in Ubuntu and ran in Chrome.

dmandrioli, 2020-06-17:
Does it work with http://sse.hup.li/edit/%2Fpointcloud_labeling.pcd ?

wuyushuwys, 2020-06-17:
Just test it on OSX. It works on OSX&safari but chrome still doesn't work. It seems that is related to the browser.

dmandrioli, 2020-06-17:
It works on OSX/Chrome on my side so I guess it's about Ubuntu/Chrome. I will investigate, thanks.

wuyushuwys, 2020-06-17:
I just found there may be some conficts with chrome extensions. after i deactivated all chrome extension it works well. Thanks!

closed 2020-06-17 by dmandrioli