Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #92: json api call returns empty

mteastepp2, 2020-06-18:
I'm having an issue using the API. I have a 2D image which has been labelled. If I go to the API in chrome (localhost:3000/api/json/(file_url)) it displays all the information for the labeling of that image. However, I can't get that to reflect in my python download script.

```
data = request.get('localhost:3000/api/json/(file_url)').json()
print('data = {}'.format(data))
```
this returns
>> data = {}

instead of what I would assume to be a dictionary with all the same information that is displayed in chrome.

as a followup, how could I then use API calls to then download the actual image with mask. I know while labeling an individual file I can press the download button in the upper right corner to download the actual image file, but I'd like to be able to do this as a batch download after labeling a lot of images.

dmandrioli, 2020-06-18:
You can try this:
```
import requests
import json
content = json.loads(requests.get("http://localhost:3000/api/json/XXX").text)
print(content)
```

Image with mask are currently generated in the browser so you can't easily script this part, I just created a feature request for this (#93)

mteastepp2, 2020-06-18:
still nothing. The status code is 200 also. what sort of debugging information could I provide that may help?

dmandrioli, 2020-06-18:
Incorrect URLs lead to empty object, you should check that URL in python exactly match the URL in the browser when you press download button.

mteastepp2, 2020-06-18:
when I'm in the actual editing tool, I've clicked on "Show JSON Output" button in the upper right. I've copied that url exactly into the python file. 
in my case, this is, "http://localhost:3000/api/json/%2fblackfly_001_00397.jpg"

dmandrioli, 2020-06-18:
Ok...weird. What's the content of  `requests.get("http://localhost:3000/api/json/XXX").headers`?

mteastepp2, 2020-06-18:
{
'content-type': 'application/json',
'vary': 'Accept-Encoding',
'date': 'Thu, 18 Jun 2020 08:07:02 GMT', 
'connection': 'keep-alive', 
'transfer-encoding': 'chunked'
}

Added newlines just to make it a little easier to read.

dmandrioli, 2020-06-18:
Looks good, what's the result using curl?
`curl http://localhost:3000/api/json/%2fblackfly_001_00397.jpg`

mteastepp2, 2020-06-18:
when ran in bash, it outputs the exact same information that's displayed in the browser. I can't paste it here, as it includes some proprietary information. If you need it for debugging though, I can photo edit out that stuff out and upload. 

I've also tried using wget in bash, and it downloads a file that when opened in a text editor displays the exact information as the browser

dmandrioli, 2020-06-18:
As a matter of fact there's something wrong with your python. No idea what the problem could be as it works on my Mac with Python 3.7.3. Good luck ;)

mteastepp2, 2020-06-18:
well that's unfortunate. Thanks for helping me though, I appreciate it! I'll keep digging around. I might just use the python curl library instead of the requests library

closed 2020-06-18 by dmandrioli

mteastepp2, 2020-06-19:
So after a ton of testing and getting some colleagues to test things for me as well, I honestly think there might be a problem with your API's. I'll just give you the details of what we did and let you decide for yourself though.

I did a brand new, fresh install of Ubuntu 20.0.4. I ran the code that you provided earlier, and still an empty dictionary. I tried every conceivable way of doing a get request. All of them returned empty, despite the API URL in my broswer displaying the label data. 
So then I tried using your get command with, 'localhost:3000/api/listing" and sure enough, that displays everything I expect it to. So '/listing' works, but '/json' doesn't when running the exact same code. 

Then I got a colleague with both a Mac and Linux machine to test it. And just as you said, the code works for '/json' on a Mac. However, he copied and pasted that code onto his Ubuntu machine, and it stopped working there. 

All in all, we've tried Python 3.6 in Ubuntu 18.04 and Python 3.8 in Ubuntu 20.04 on one machine, and Python 3.8 in Ubuntu 20.04 on a second machine. None of them worked. Not sure what python version was ran on the Mac. 

So it seems to me that for some reason, the '/json' API just doesn't want to work with Ubuntu. Both the '/listing' and '/pcdfile' API's work just fine for us though. To get around this, I decided to just do an os.system() call to run a wget command to download the json's we needed. I'm really not sure if the source of the problem is Ubuntu or the way you have setup the API though. Regardless, thank you for making this tool. It's still making our work 100 times easier.

dmandrioli, 2020-06-19:
Thanks for your detailed feedback, I tried to reproduce using a fresh Ubuntu virtual machine but it works, at least with the sample file. You can look at [this video](https://github.com/dmandrioli/sse-extra/raw/master/sseubuntu.mp4) to see what I tried. Maybe your issue is related to some specific character(s) in your file names that would need to be escaped in some ways?

mteastepp2, 2020-06-19:
I'm absolutely baffled. I really don't know why it's not working for me. But it clearly seems to be an issue on my end then. The wget solution is working so far, but if we run into issues with that solution then I'll investigate later. Only primary difference I noticed is your video uses a png file while we're using a jpg, and then the actual file name. But I highly doubt that's the cause. Thanks so much again for helping me look into this! I appreciate the time and effort you put in.

dmandrioli, 2020-06-19:
You're welcome. As a freelance developer I can provide professional support and custom development around this tool, saving your and your colleagues valuable time if needed ;)