Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #95: Maximum PCD point count support

aidonchuk, 2020-06-23:
Hi! Do you have any maximum points limit for PCD files?

dmandrioli, 2020-06-23:
Hi, no you are only limited by the power of your CPU and GPU.

aidonchuk, 2020-06-23:
May I give you 2 files that I cut from one big file by the volume of X coords?(booth they have same width)
The problem that big file not opening at all but small is ok.
My PC AMD 3950, 32gb, 2x1080ti, Ubuntu 18

dmandrioli, 2020-06-23:
Ok send these files to see their characteristics.

aidonchuk, 2020-06-23:
Thanks a lot! 
https://yadi.sk/d/OciOXUeccBwlYw
This link for file *_0.pcd - opening lasts forever and *_4.pcd - ok.

dmandrioli, 2020-06-23:
I just tried to load the 2 files, it works as expected. No idea of what's going on for you but it can't be related to the number of points. The bigger file is about 60k points which is low for the tool even with a slow machine. You may try other browsers/OS to double check.

aidonchuk, 2020-06-23:
Thanks for help and sorry for the disturbance.

closed 2020-06-23 by aidonchuk

dmandrioli, 2020-06-23:
No problem, if you find a solution feel free to share it here.