Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #96: Descriptions for Dockerfile folder parameters

mannatmkm, 2020-06-24:
Thank you for this labeling tool! 
I'm trying to generate a docker image so that I can modify the tool.
In the given dockerfile, there are a few parameters that I'm unclear about. $APP_SOURCE_FOLDER, I think refers to this semantic-segmentation editor folder. But what relates to $SCRIPTS_FOLDER and $APP_BUNDLE_FOLDER, and where can I find the corresponding shell scripts for building meteor, npm, entrypoint, etc.

dmandrioli, 2020-06-24:
Thank you! To modify the tool, running it with Docker is not a good idea, you will have to wait long time for the building process.
You should prefer running the tool from source using `meteor --settings settings.json`

mannatmkm, 2020-06-24:
Thank you. I'll try that.

closed 2020-06-24 by dmandrioli