Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #97: Demo on the main page for the bitmap editor doesn't work

urishab, 2020-07-02:
Giving me the following error:

Your app is crashing. Here's the latest log:
Semantic Segmentation Editor
Images (JPG, PNG, PCD) served from /home/ssedemo/sse-images
PCD binary segmentation data stored in /home/ssedemo/sse-internal
Number of available sets of object classes: 3
=> Meteor server restarted
meteor://ðŸ’»app/app/app.js:438
  if (bytes.byteLength > 0) {
            ^

TypeError: Cannot read property 'byteLength' of undefined
    at bytesToObject (server/SseDataWorkerServer.js:198:15)
    at Object.uncompress (server/SseDataWorkerServer.js:207:26)
    at ReadFileContext.readFile (server/api.js:88:48)
    at FSReqWrap.readFileAfterOpen [as oncomplete] (fs.js:420:13)
Exited with code: 1
Your application is crashing. Waiting for file change.
=> A patch (Meteor 1.8.1) for your current release is available!
   Update this project now with 'meteor update --patch'.
=> A patch (Meteor 1.8.1) for your current release is available!
   Update this project now with 'meteor update --patch'.

urishab, 2020-07-02:
direct link to the demo:
http://sse.hup.li/edit/%2Fbitmap_labeling.png

dmandrioli, 2020-07-02:
Thanks for reporting, it works now.

closed 2020-07-02 by dmandrioli

urishab, 2020-07-02:
THanks! That's service :)