Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #99: Questions about pcd point cloud loading

baifanxxx, 2020-07-15:
Hello author, 
Thank you very much for designing this tool, it greatly helped our work. However, I encountered the problem of loading point cloud files. I successfully configured this tool on the Windows 10 system, but when I load the point cloud file in pcd format in the webpage, the webpage has been waiting and cannot be loaded successfully. When I try to load a jpg format image, the loading speed is very fast and I can edit it directly. I don't understand what caused this. My computer is equipped with 1050TI GPU and 4G memory. The point cloud file loaded is only Stanford's rabbit, not big. Hope you can help me solve this problem, thank you very much!

dmandrioli, 2020-07-18:
Hi, to investigate what's going on, can you share "Stanford's rabbit" you are talking about?

baifanxxx, 2020-07-19:
Hi,
Thanks for your help. I have solved this problem. Different browsers will affect the display of point cloud.

closed 2020-07-19 by dmandrioli