Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #100: Move through the scene with WASD

DarioSiroki, 2020-07-16:
Would you accept if I made a pull request to move through the scene with WASD? I'm talking about X and Z axis. It's a simple addition which only requires some configuration of [OrbitControls](https://threejs.org/docs/#examples/en/controls/OrbitControls).

And a question for you, do you have any idea what would be the easiest way to add Q/E keys to change height (Y axis)? I'm not too experienced with Three.js, but it looks like switching to any other controlling model would be too complex or impossible. My conclusion is that the best option would be to edit the OrbitControls to make this possible?

dmandrioli, 2020-07-18:
Hi, you are free to submit pull requests and you are right, improving navigation can't be achieved by modifying OrbitControls.

DarioSiroki, 2020-07-19:
Thanks, I will make a pull request for height navigation too maybe if I will need it or if I will have time

dmandrioli, 2020-07-20:
Fixed in 1.5.2

closed 2020-07-20 by dmandrioli