Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #103: Modifications to the application

mannatmkm, 2020-07-21:
Hey Damien, 
I'd appreciate if you could tell me which scripts and what to change for the following modifications for the 3D editor:
1) No zooming out after double clicks
2) Adding/changing object instance id in the editor
3) After deletion of objects, to delete the labels as well. (I think that's already a reported bug)
4) Adding a paintbrush option to paint the point cloud
5) Having a review mode, where one can change the labels and then view the comparisons from original, to provide feedback.

dmandrioli, 2020-07-21:
Hi Mannat,
1. Zooming out is not triggered by double click but by clicking in the background. (see `SseEditor3d.mouseUp` method)
2. The first thing would be to add an id to objects which is not the case for now. 
3.  You're right, this is a registered bug. The method to fix is probably `SseEditor3d.deleteSelectedObject`
4. You would have to create a new tool. You can start from existing ones in `/imports/editor/3d/tools/*`
5. Big feature, would be too long to elaborate.

Hope this helps

mannatmkm, 2020-07-21:
It did help a bit, Thank you.

closed 2020-08-09 by dmandrioli