Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #109: How to choose the pcd file to annotate

EricWiener, 2020-10-28:
Thank you for making this great tool available for use! I wasn't able to figure out how to choose what PCD file to annotate. I'm sure I'm just overlooking something simple, but I would really appreciate a pointer.

I configured the `settings.json` to have:
```
"configuration": {
    "images-folder": "",
    "internal-folder": "/Users/ericwiener/Downloads/pcd"
  },
``` 
And I put my pcd file in the `/Users/ericwiener/Downloads/pcd` directory. When I load the tool, it adds a new file `pointcloud_labeling.pcd.labels` to the directory, which makes me think the directory is detected correctly. Additionally, I get the message `I20201028-19:00:11.938(-4)? PCD binary segmentation data stored in /Users/ericwiener/Downloads/pcd` in the logs, which again makes me think the directory is detected correctly.

Once I'm in the tool, I can't figure out how to choose a different pcd file than the default one the tool comes with.

Thanks so much!

dmandrioli, 2020-10-28:
Thanks Eric!
"images-folder" is where jpg and pcd files are stored, try this
```
"configuration": {
    "images-folder": "/Users/ericwiener/Downloads/pcd",
    "internal-folder": ""
  },
```
 Doc: https://github.com/Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor#configuration-file-settingsjson

EricWiener, 2020-10-28:
Thanks so much @dmandrioli! That worked perfectly and I'm able to load the pcd files. However, if I try to load more than 20 and browse to the next page of pcd files, I get the error `/Users/ericwiener/Downloads/pcd/null does not exist. Check the documentation and your settings`.

dmandrioli, 2020-10-28:
This is a bug. As a workaround you can edit the URL manually. The pattern is /browse/PAGE_INDEX/ITEMS_COUNT
So you can try /browse/1/20 or /browse/0/200

Feel free to create a bug issue for this one, thanks for reporting.

EricWiener, 2020-10-28:
Got it. I will create a new issue for this. One last question, is there a specific formatting the PCD file should be in? I have two different PCD formats (one from a Velodyne and one exported from an annotation tool). The one exported from the annotation tool shows up fine. The one from the Velodyne seems to be aligned just along the XYZ axis.
[image]
[image]

EricWiener, 2020-10-28:
#110 Created a new issue for the bug

dmandrioli, 2020-10-28:
You can try to change the global orientation using the bottom right button (cycling arrows, not very explicit...)

EricWiener, 2020-10-28:
Unfortunately, that just rotates the XYZ coordinates but doesn't display the points correctly. The file is XYZIR point clouds (XYZ, intensity, ring # from the Velodyne). Is this supported or should I convert to a different format?

dmandrioli, 2020-10-28:
It is supported as long as you have XYZ columns. If you share the badly displayed pcd file, I can investigate in the next days.

EricWiener, 2020-10-29:
Thank you! Here's the badly displayed pcd file. I uploaded it as a zip of the single file since GitHub doesn't support uploading .pcd files.
[1561043068.656119000.pcd.zip](https://github.com/Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor/files/5455388/1561043068.656119000.pcd.zip)

EricWiener, 2020-11-04:
Hi @dmandrioli. Did you have a chance to look at the PCD file? In case it helps, here is the header of a file that correctly displayed for me:
```
# .PCD v0.7 - Point Cloud Data file format
VERSION 0.7
FIELDS x y z rgba label
SIZE 4 4 4 4 4
TYPE F F F U U
COUNT 1 1 1 1 1
WIDTH 268
HEIGHT 1
VIEWPOINT 0 0 0 1 0 0 0
POINTS 268
DATA binary
```
Here is one that didn't work:
```
# .PCD v0.7 - Point Cloud Data file format
VERSION 0.7
FIELDS x y z _ intensity ring _
SIZE 4 4 4 1 4 2 1
TYPE F F F U F U U
COUNT 1 1 1 4 1 1 10
WIDTH 330
HEIGHT 1
VIEWPOINT 0 0 0 1 0 0 0
POINTS 330
DATA binary
```

dmandrioli, 2020-11-06:
Sorry @EricWiener, very busy these days... I didn't dive into this but looking again your badly rendered PCD file in the screenshot, the problem looks to be more in original data projection than in the tool. The overall shape looks like the origin of a 3d coordinate system.

EricWiener, 2020-11-07:
@dmandrioli got it - no worries. I really appreciate any help. I agree about it looking like the 3d coordinate system. I'm going to try to see if converting the PCD file to a different format helps.