Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #114: cannot run!

daixiaolei623, 2020-12-09:
Thank you for your code.
But when i run the 'meteor npm start', i got the errors:
'npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! semantic-segmentation-editor@1.2.0 start: `meteor --settings settings.json --production`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the semantic-segmentation-editor@1.2.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-12-09T09_40_44_366Z-debug.log'
Could you please tell me what is the problem and how to run.
Thank you.

dmandrioli, 2020-12-09:
Hi, could you copy/paste here the content of /root/.npm/_logs/2020-12-09T09_40_44_366Z-debug.log?

closed 2020-12-25 by dmandrioli

MinglinChen94, 2021-05-10:
Hi, I met the same problem.

And the .log file as follows:

0 info it worked if it ends with ok
1 verbose cli [
1 verbose cli   '/home/mlchen/.meteor/packages/meteor-tool/.2.2.0.1j8auib.qcbe++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/dev_bundle/bin/node',
1 verbose cli   '/home/mlchen/.meteor/packages/meteor-tool/.2.2.0.1j8auib.qcbe++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/dev_bundle/bin/npm',
1 verbose cli   'start'
1 verbose cli ]
2 info using npm@6.14.12
3 info using node@v12.22.1
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle semantic-segmentation-editor@1.6.0~prestart: semantic-segmentation-editor@1.6.0
6 info lifecycle semantic-segmentation-editor@1.6.0~start: semantic-segmentation-editor@1.6.0
7 verbose lifecycle semantic-segmentation-editor@1.6.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle semantic-segmentation-editor@1.6.0~start: PATH: /home/mlchen/.meteor/packages/meteor-tool/.2.2.0.1j8auib.qcbe++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/dev_bundle/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/mlchen/code/SemanticSegmentationEditor/semantic-segmentation-editor-1.6.0/node_modules/.bin:/home/mlchen/.meteor/packages/meteor-tool/.2.2.0.1j8auib.qcbe++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/dev_bundle/bin:/home/mlchen/.meteor/packages/meteor-tool/.2.2.0.1j8auib.qcbe++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64:/home/mlchen/.meteor/packages/meteor-tool/.2.2.0.1j8auib.qcbe++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/dev_bundle/lib/node_modules/.bin:/home/mlchen/anaconda3/condabin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
9 verbose lifecycle semantic-segmentation-editor@1.6.0~start: CWD: /home/mlchen/code/SemanticSegmentationEditor/semantic-segmentation-editor-1.6.0
10 silly lifecycle semantic-segmentation-editor@1.6.0~start: Args: [
10 silly lifecycle   '-c',
10 silly lifecycle   'meteor run --settings settings.json --exclude-archs "web.browser.legacy, web.cordova"'
10 silly lifecycle ]
11 silly lifecycle semantic-segmentation-editor@1.6.0~start: Returned: code: 1  signal: null
12 info lifecycle semantic-segmentation-editor@1.6.0~start: Failed to exec start script
13 verbose stack Error: semantic-segmentation-editor@1.6.0 start: `meteor run --settings settings.json --exclude-archs "web.browser.legacy, web.cordova"`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/home/mlchen/.meteor/packages/meteor-tool/.2.2.0.1j8auib.qcbe++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/dev_bundle/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:314:20)
13 verbose stack     at ChildProcess.<anonymous> (/home/mlchen/.meteor/packages/meteor-tool/.2.2.0.1j8auib.qcbe++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/dev_bundle/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:314:20)
13 verbose stack     at maybeClose (internal/child_process.js:1022:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:287:5)
14 verbose pkgid semantic-segmentation-editor@1.6.0
15 verbose cwd /home/mlchen/code/SemanticSegmentationEditor/semantic-segmentation-editor-1.6.0
16 verbose Linux 5.4.0-72-generic
17 verbose argv "/home/mlchen/.meteor/packages/meteor-tool/.2.2.0.1j8auib.qcbe++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/dev_bundle/bin/node" "/home/mlchen/.meteor/packages/meteor-tool/.2.2.0.1j8auib.qcbe++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/dev_bundle/bin/npm" "start"
18 verbose node v12.22.1
19 verbose npm  v6.14.12
20 error code ELIFECYCLE
21 error errno 1
22 error semantic-segmentation-editor@1.6.0 start: `meteor run --settings settings.json --exclude-archs "web.browser.legacy, web.cordova"`
22 error Exit status 1
23 error Failed at the semantic-segmentation-editor@1.6.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

MinglinChen94, 2021-05-10:
Hi, I met the same problem.

The .log file are as follows:

0 info it worked if it ends with ok
1 verbose cli [
1 verbose cli   '/home/mlchen/.meteor/packages/meteor-tool/.2.2.0.1j8auib.qcbe++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/dev_bundle/bin/node',
1 verbose cli   '/home/mlchen/.meteor/packages/meteor-tool/.2.2.0.1j8auib.qcbe++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/dev_bundle/bin/npm',
1 verbose cli   'start'
1 verbose cli ]
2 info using npm@6.14.12
3 info using node@v12.22.1
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle semantic-segmentation-editor@1.6.0\~prestart: semantic-segmentation-editor@1.6.0
6 info lifecycle semantic-segmentation-editor@1.6.0\~start: semantic-segmentation-editor@1.6.0
7 verbose lifecycle semantic-segmentation-editor@1.6.0\~start: unsafe-perm in lifecycle true
8 verbose lifecycle semantic-segmentation-editor@1.6.0\~start: PATH: /home/mlchen/.meteor/packages/meteor-tool/.2.2.0.1j8auib.qcbe++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/dev_bundle/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/mlchen/code/SemanticSegmentationEditor/semantic-segmentation-editor-1.6.0/node_modules/.bin:/home/mlchen/.meteor/packages/meteor-tool/.2.2.0.1j8auib.qcbe++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/dev_bundle/bin:/home/mlchen/.meteor/packages/meteor-tool/.2.2.0.1j8auib.qcbe++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64:/home/mlchen/.meteor/packages/meteor-tool/.2.2.0.1j8auib.qcbe++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/dev_bundle/lib/node_modules/.bin:/home/mlchen/anaconda3/condabin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
9 verbose lifecycle semantic-segmentation-editor@1.6.0\~start: CWD: /home/mlchen/code/SemanticSegmentationEditor/semantic-segmentation-editor-1.6.0
10 silly lifecycle semantic-segmentation-editor@1.6.0\~start: Args: [
10 silly lifecycle   '-c',
10 silly lifecycle   'meteor run --settings settings.json --exclude-archs "web.browser.legacy, web.cordova"'
10 silly lifecycle ]
11 silly lifecycle semantic-segmentation-editor@1.6.0\~start: Returned: code: 1  signal: null
12 info lifecycle semantic-segmentation-editor@1.6.0\~start: Failed to exec start script
13 verbose stack Error: semantic-segmentation-editor@1.6.0 start: `meteor run --settings settings.json --exclude-archs "web.browser.legacy, web.cordova"`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/home/mlchen/.meteor/packages/meteor-tool/.2.2.0.1j8auib.qcbe++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/dev_bundle/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:314:20)
13 verbose stack     at ChildProcess.<anonymous> (/home/mlchen/.meteor/packages/meteor-tool/.2.2.0.1j8auib.qcbe++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/dev_bundle/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:314:20)
13 verbose stack     at maybeClose (internal/child_process.js:1022:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:287:5)
14 verbose pkgid semantic-segmentation-editor@1.6.0
15 verbose cwd /home/mlchen/code/SemanticSegmentationEditor/semantic-segmentation-editor-1.6.0
16 verbose Linux 5.4.0-72-generic
17 verbose argv "/home/mlchen/.meteor/packages/meteor-tool/.2.2.0.1j8auib.qcbe++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/dev_bundle/bin/node" "/home/mlchen/.meteor/packages/meteor-tool/.2.2.0.1j8auib.qcbe++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/dev_bundle/bin/npm" "start"
18 verbose node v12.22.1
19 verbose npm  v6.14.12
20 error code ELIFECYCLE
21 error errno 1
22 error semantic-segmentation-editor@1.6.0 start: `meteor run --settings settings.json --exclude-archs "web.browser.legacy, web.cordova"`
22 error Exit status 1
23 error Failed at the semantic-segmentation-editor@1.6.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

odddozen, 2021-08-11:
Same had same/similar issue - with multiple different built releases (1.6, 1.5.3, 1.5.2).  Even more odd - I had 1.5.3 working and then the problem cropped up again.  Heres a debug log for me.
[2021-08-11T02_10_01_231Z-debug.log](https://github.com/Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor/files/6965433/2021-08-11T02_10_01_231Z-debug.log)