Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #116: pcd files created from Ouster sensor data do not work

9598jhj, 2020-12-15:
I have two sensor such as vlp-16 and ouster-64
The sensor data made from vlp-16 works fine with this code, but
Sensor data made form ouster does not worked Is there a solution to this problem?

dmandrioli, 2021-01-04:
Can you share a PCD file not supported by the tool? Thanks.

283pm, 2021-05-12:
I have the same issue with ouster-128.
This is one of pcd files made from rosbag file.
Are there any problem on this file?
Thanks

[1620798655.374639156.zip](https://github.com/Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor/files/6464473/1620798655.374639156.zip)

dmandrioli, 2021-05-12:
Hi, the tool decode the shape of your data as: 262144 points /  Width: 6.7590201069809344e+38m / Height: 6.757695868457912e+38m / Depth: 6.727824746890038e+38m

The number of points is correct but coordinates are not. You can tweak SsePCDLoader.js to investigate what is going on or you may want to convert your PCD file to the ASCII format to see if it looks better.

283pm, 2021-05-13:
Thank you for replying.
I tested to convert binary PCD files to the ASCII format, and this tool works fine.  Thanks for a great tool.