Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #117: What is the data type of the label file?

hamin-song, 2020-12-17:
Hi, thanks for making awesome labeling tool!!

I'm going to use your annotation tool as auto label correction.
So I want to import label files which were predicted from deep learning model (such as RangeNet++, ConvPoint ...)
And I'm trying to convert prediction results into the format of label file that your tool creates, 
but I'm having a hard time because I don't know the format.

What is the **data type of the label file** created in sse-internal?

dmandrioli, 2020-12-17:
Hi, thank you 👍 

Are you labeling bitmap images or point clouds?

hamin-song, 2020-12-17:
> Hi, thank you 👍
> 
> Are you labeling bitmap images or point clouds?

Wow.. thank you for your quick reply !! :-)
I'm labeling point clouds!!

dmandrioli, 2020-12-17:
The format is a JSON descriptor compressed with 2 different algorithms in a web worker to avoid browser hanging. The descriptor format is not documented but you may be able to look at it by adding breakpoints into [this file](https://github.com/Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor/blob/master/imports/common/SseDataManager.js) which is responsible for the encoding of the output format.

By the way importing external data is not supported by the tool, it could be non trivial to implement.

Hope this helps.

hamin-song, 2020-12-17:
> The format is a JSON descriptor compressed with 2 different algorithms in a web worker to avoid browser hanging. The descriptor format is not documented but you may be able to look at it by adding [...]

Thank you very much for these information!!
I don't think it'll be easy, but I'll try.
Thanks 👍

GXU-doudou, 2021-12-30:
@hamin-song hello, did you deal with this issue? I have the same problem with you. Hope you can reply，thanks!