Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #121: Intensity value in Viewing and Exported .pcd

Benjamin-Tan, 2021-02-24:
Thanks for such an amazing editor tool. Is it possible to have the above-mentioned feature to be merged into the main release? I have seen an attempt in #53 but it has been closed but not merged to the main release.

dmandrioli, 2021-02-24:
Thank you :)
The mentioned PR has not been merged but a support for Intensity display has been implemented in 1.4.0 (https://github.com/Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor/compare/1.3.0...1.4.0)
Hope this helps.

dmandrioli, 2021-02-24:
Hum sorry in fact this is about RGB not intensity.

dmandrioli, 2021-02-24:
I remember the story now. Supporting RGB point clouds lets you encode intensity as RGB. So it's up to you to provide RGB values to represent intensity. Thanks.

Benjamin-Tan, 2021-02-24:
thanks for the speedy reply. So you would recommend to use RGB to represent the intensity value instead of having XYZIRGB?

dmandrioli, 2021-02-24:
Yes, this is what is supported by the tool.

closed 2021-02-25 by Benjamin-Tan