Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #126: .pcd file can not load

zx0815, 2021-03-16:
Hello,
I installed the latest version 1.6.0 and meteor2.1.
But I can hardly open any .pcd file, the only file I can open is a pcd file with horses, but it only opens a part of it, and I can’t perform other operations except for the right-click box selection (such as rotation) . Focus selection, etc. I have another question. The bottom right corner of the webpage always displays "loading objects..." or "loading PCD data..." or "loading label...". I have tried to open more than ten PCD files. ,feeling bad. The picture below is the picture displayed when I load the horse. Obviously he only loaded part of it, but I really can’t operate it.
[image]

dmandrioli, 2021-03-16:
Hi, can you open the default pcd sample?
Did you updated meteor with the command "meteor update" or do you stick to the bundled version of meteor in 1.6.0?

zx0815, 2021-03-16:
> 嗨，您可以打开默认的pcd示例吗？
> 您是否使用命令“ meteor update”更新了流星，还是坚持使用1.6.0的捆绑版流星？

Do you mean whether "DEMO: Bitmap editor" can be opened. I can open and operate this, but when I click the button in the upper right corner to download the pcd file, it always fails, and I can only download a 1kb file in the end. I used the "meteo update" command to update to the latest version.

dmandrioli, 2021-03-16:
In the online demo, downloading is disabled. I'm talking about opening pointcloud_labeling.pcd in the tool you run locally.
Updating meteor should not be a problem but I didn't test yet.

zx0815, 2021-03-16:
> 在在线演示中，下载被禁用。我说的是在本地运行的工具中打开pointcloud_labeling.pcd。
> 更新流星应该不是问题，但是我还没有测试。

> In the online demo, downloading is disabled. I'm talking about opening pointcloud_labeling.pcd in the tool you run locally.
> Updating meteor should not be a problem but I didn't test yet.

Okay, I understand, I just put the pointcloud_labeling.pcd file into the sse-images folder, and it still doesn’t load it
[image]

zx0815, 2021-03-17:
By the way, my computer is a Win10 computer