Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #128: Can't visualize .pcd files in webpage

Oscarstenstad, 2021-03-26:
Hi, 
I have been trying to use this software on .pcd files and everything seems to be working fine until I get in to the actual Segmentation page of the webpage. I can choose a .pcd file from the directory I have chosen but the image doesn't show in the next page. Anyone know how to fix this? I have tried with both meteor 1.6 and 2.1 and neither work.
[image]

dmandrioli, 2021-03-26:
Hi, does the sample pcd file work?
If not, could you please share your problematic pcd file?
Thanks.

Oscarstenstad, 2021-03-26:
Hi again, 
I can open the bitmap file, but the sample pcd does not work either, it just says "loading objects..." in the bottom right corner, as it did when I tried with my own file. I have let it load for a really long time but nothing happens. 

The header of the file I am trying to open looks like this:
[image]

And here is a .txt file with the content of the pcd file
[txt_cloud.txt](https://github.com/Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor/files/6211192/txt_cloud.txt)

dmandrioli, 2021-03-30:
Hi, could you please provide the original PCD file which is in binary format according to headers.
Thanks.

jobberz77, 2021-03-31:
I had trouble with displaying the sample pointcloud and others aswell in the newest version. For some reason going back to version 1.5.2 fixed that for me

Oscarstenstad, 2021-04-06:
I tried your suggestion and it works here as well. Thank you for the suggestion :)

Atlaszjr-star, 2021-07-04:
> I had trouble with displaying the sample pointcloud and others aswell in the newest version. For some reason going back to version 1.5.2 fixed that for me

Hi @jobberz77 and @Oscarstenstad  I am new to Meteor and this software, may I ask how you used the 1.5.2 version to run this software? this software seems to built by Meteor@1.12 version. Thank you in advance!

jobberz77, 2021-07-04:
Hello @Atlaszjr-star, I was referring to version 1.5.2 of the labeling tool, not the Meteor version. You can get the 1.5.2 version of the labeling tool by downloading the branch of that version. You don't need to change the Meteor version.
Let me know if you need more help!

Atlaszjr-star, 2021-07-04:
Hi @jobberz77 , thank you so much for your advice, I just realized it's the labeling tool's version ;-) it does work right now!
Thanks again!

xipi702, 2021-10-17:
@robertson-oa 
Hi, bro, would you like to help me? how did you solve it? using version 1.5.2?

amansth, 2021-11-05:
@jobberz77 I'm having the same problem with Cloudcompare generated file. Did you get it working by changing the data to ASCII and changing the headers ? 

I'm trying the latter but I'm not sure what the _ field should be.

Current header that doesn't work: 
# .PCD v0.7 - Point Cloud Data file format
VERSION 0.7
FIELDS Classification rgb x y z _
SIZE 4 4 4 4 4 1
TYPE F F F F F U
COUNT 1 1 1 1 1 4
WIDTH 505622
HEIGHT 1
VIEWPOINT 0 0 0 1 0 0 0
POINTS 505622
DATA binary

Header, which works: 
# .PCD v.7 - Point Cloud Data file format
VERSION .7
FIELDS x y z rgb
SIZE 4 4 4 4
TYPE F F F F
COUNT 1 1 1 1
WIDTH 213
HEIGHT 1
VIEWPOINT 0 0 0 1 0 0 0
POINTS 213
DATA ascii

amansth, 2021-11-05:
Actually never mind, I just managed to fix it. 

The solution for me was as @jobberz77 suggested: use open3d. 

For anyone still stuck and found their way here,
the functions of interest are: 
open3d.io.read_point_cloud                             - to open pcd file with incorrect header from cloudcompare
open3d.visualisation.draw_geometries             - to verify that the model looks as it should
open3d.io.write_point_cloud                             - to write in the correct format.

pip3 install open3d , then you can use this small script: 

import numpy as np
from open3d import *

def main():
            cloud = open3d.io.read_point_cloud("path to pcd")
            open3d.visualization.draw_geometries([cloud])
            open3d.io.write_point_cloud("path for pcd")

if __name__== "__main__":
            main()

wsn, 2021-12-28:
@jobberz77  I still can't read sample .pcd file through 1.5.2  tool and meteor 2.5.1, the result is following and another problem is that when run start ,it always report "Browserslist: caniuse-lite is outdated".  It would be much appreciated if anyone could help me.
[image]

jobberz77, 2021-12-28:
@wsn I have seen that caniuse-lite error before and I don't think you should worry about it too much. For the black screen you are getting though, its different from the error I had which is an infinite loading loop when using the wrong type of .pcd files. 

Honestly, I'm not sure if I can help you with this. Maybe changing up the Meteor version fixes the problem but I can't say for sure. Good luck!