Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #130: Layers are sometimes reset suddenly

kotetsu-n, 2021-04-02:
Hi, thank you for developing such a great tool. But I have encountered a problem. As I have written in title, the layers which already set for each label sometimes can be reset suddenly (reset means to go to background layer). Do you know why it happens? Also, can I change the layer of multiple labels at the same time? 
Thanks,

dmandrioli, 2021-04-02:
Thanks!
Unfortunately it's difficult to figure out what's going on about the reset of your labelling layer.
For your second question, you can't change the layer of multiple labels at the same time because it's not possible to select multiple polygons.

kotetsu-n, 2021-04-02:
Thanks, I understand. In terms of my first question, I could reproduce it. When I want to add a new label on/around the labels using already exist using rectangle tool or polygon tool, it sometimes results to add some vertices to the label already exists. At that time, the number of the object on the status bar doesn't increase but the number of the points increases. Then, at that time, if you hit Undo button, all labels go to background layer. If you can reproduce it, it would be great if you could modify it.

dmandrioli, 2021-04-02:
Thanks for this clear explanation, I just reproduced your problem easily. And I already know how I will fix it :)
I will disable undo/redo during creation of a new polygon as it does not really make sense. Actually to remove the last point(s) of a polygon you are creating, you can press ESC.
Hope this helps and thanks for reporting.

kotetsu-n, 2021-04-02:
I see, thanks!

dmandrioli, 2021-04-03:
Reopening to track the bug fix.