Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #131: TypeError: Cannot read property 'demo-mode' of undefined

KernelA, 2021-04-19:
I am trying to run a demo in Docker, but I get the following error:
```
app_1    | Semantic Segmentation Editor
app_1    | Images (JPG, PNG, PCD) served from /root/sse-images
app_1    | PCD binary segmentation data stored in /root/sse-internal
app_1    | Number of available sets of object classes: 1
app_1    | /opt/bundle/bundle/programs/server/node_modules/fibers/future.js:280
app_1    |                                              throw(ex);
app_1    |                                              ^
app_1    |
app_1    | TypeError: Cannot read property 'demo-mode' of undefined
app_1    |     at module (server/files.js:8:18)
app_1    |     at fileEvaluate (packages/modules-runtime.js:336:7)
app_1    |     at Module.require (packages/modules-runtime.js:238:14)
app_1    |     at require (packages/modules-runtime.js:258:21)
app_1    |     at /opt/bundle/bundle/programs/server/app/app.js:1376:1
app_1    |     at /opt/bundle/bundle/programs/server/boot.js:401:38
app_1    |     at Array.forEach (<anonymous>)
app_1    |     at /opt/bundle/bundle/programs/server/boot.js:226:21
app_1    |     at /opt/bundle/bundle/programs/server/boot.js:464:7
app_1    |     at Function.run (/opt/bundle/bundle/programs/server/profile.js:280:14)
app_1    |     at /opt/bundle/bundle/programs/server/boot.js:463:13
mongo_1  | {"t":{"$date":"2021-04-19T11:58:45.496+00:00"},"s":"I",  "c":"NETWORK",  "id":22944,   "ctx":"conn2","msg":"Connection ended","attr":{"remote":"172.20.0.3:34796","connectionId":2,"connectionCount":0}}
```

Docker on Windows 10 Pro 2004:
```
docker --version
Docker version 20.10.5, build 55c4c88
```

Project tag version: 1.6.0
```
git rev-parse HEAD
9e7a16b8f69570fa65d4e219db9e286c2ce47961
```

Jensssen, 2021-04-27:
I ran into the same issue.
OS: Ubuntu 18.04
Docker: Docker version 20.10.6, build 370c289

dmandrioli, 2021-04-27:
Thanks for reporting. The quick fix rely on instructions update.

commit b4523bb8 (2021-04-27): Update README.md

Update documentation (fix #131)
closed 2021-04-27 by dmandrioli