Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #134: Errors starting app (METEOR_SETTINGS are not valid JSON; Cannot read property 'demo-mode' of undefined)

adrianstrkr, 2021-05-19:
Hello,

i´d really like to use your app in a scientific context. However, unfortunatelly I have plenty of issues setting it up. 
In order to simplyfy the starting process of your app I created an .env file in which I specify the env variables "METEOR_SETTINGS" and  "SSE_IMAGES" (see below). Herby I set the path to the point clouds I´d like to work with and the path to the setting json file you are providing. Note that I did not conduct any changes within the settings json file. Further no changes have been conducted in the sse-docker-stat.yml file.
.env file content:
`SSE_IMAGES=C:\sse_pc`
`METEOR_SETTINGS=.\settings.json`

Then I start the app using the following cmd prompt:
`docker-compose --env-file SSE_IMAGES.env  -f sse-docker-stack.yml up`

Consequently the mongo db is starting and a connection is successfully established to it. However while the sse app is automatically started I get the following error message: 

"app_1    | Starting app...
app_1    | /opt/bundle/bundle/programs/server/node_modules/fibers/fibers.js:90
app_1    |                                      return fn.apply(this, arguments);
app_1    |                                                ^
app_1    |
app_1    | Error: METEOR_SETTINGS are not valid JSON.
app_1    |     at packages/meteor.js:63:11
app_1    |     at packages/meteor.js:88:4
app_1    |     at packages/meteor.js:1508:3
app_1    |     at /opt/bundle/bundle/programs/server/boot.js:401:38
app_1    |     at Array.forEach (<anonymous>)
app_1    |     at /opt/bundle/bundle/programs/server/boot.js:226:21
app_1    |     at /opt/bundle/bundle/programs/server/boot.js:464:7
app_1    |     at Function.run (/opt/bundle/bundle/programs/server/profile.js:280:14)
app_1    |     at /opt/bundle/bundle/programs/server/boot.js:463:13
semantic-segmentation-editor-master_app_1 exited with code 1"

Afterwards the mongodb is still running. I really hope you can help solving this issue! Do I need to apply changes in the settings.json file. Or is something wrong with the way I set the variables in the .env file?

Further I tried to not specify the "METEOR_SETTINGS" variables at all. However, this approach resulted in the following error message: 

app_1    | Semantic Segmentation Editor
app_1    | Images (JPG, PNG, PCD) served from /root/sse-images
app_1    | PCD binary segmentation data stored in /root/sse-internal
app_1    | Number of available sets of object classes: 1
app_1    | /opt/bundle/bundle/programs/server/node_modules/fibers/future.js:280
app_1    |                                              throw(ex);
app_1    |                                              ^
app_1    |
app_1    | TypeError: Cannot read property 'demo-mode' of undefined
app_1    |     at module (server/files.js:8:18)
app_1    |     at fileEvaluate (packages/modules-runtime.js:336:7)
app_1    |     at Module.require (packages/modules-runtime.js:238:14)
app_1    |     at require (packages/modules-runtime.js:258:21)
app_1    |     at /opt/bundle/bundle/programs/server/app/app.js:1376:1
app_1    |     at /opt/bundle/bundle/programs/server/boot.js:401:38
app_1    |     at Array.forEach (<anonymous>)
app_1    |     at /opt/bundle/bundle/programs/server/boot.js:226:21
app_1    |     at /opt/bundle/bundle/programs/server/boot.js:464:7
app_1    |     at Function.run (/opt/bundle/bundle/programs/server/profile.js:280:14)
app_1    |     at /opt/bundle/bundle/programs/server/boot.js:463:13
mongo_1  | {"t":{"$date":"2021-05-19T12:24:19.783+00:00"},"s":"I",  "c":"NETWORK",  "id":22944,   "ctx":"conn2","msg":"Connection ended","attr":{"remote":"172.18.0.3:36068","connectionId":2,"connectionCount":0}}
semantic-segmentation-editor-master_app_1 exited with code 1

Can you explain how it comes to this error message and how to solve the issue?

My specs:
Windows 10 64 bit
docker-compose version 1.28.5, build c4eb3a1f
docker-py version: 4.4.4
CPython version: 3.9.0
OpenSSL version: OpenSSL 1.1.1g  21 Apr 2020

Thank you very much in advance!

best regards,
Adrian

adrianstrkr, 2021-05-19:
Hi,

thank you allot for your very fast reply! Unfortunately I did not find a equivalent prompt to cat under windows. So I guess I´ll set up the app using the linux bash shell. Hope that´ll do. If not, it should work on the latest stand alone Ubuntu distribution shouldn´t it?
If I make any progress or encounter any other obstacles I´ll let you know!

best wishes,
Adrian

dmandrioli, 2021-05-19:
Linux is for sure the best choice to run the tool. But if you successfully run it on Windows, feel free to share your remarks.

adrianstrkr, 2021-05-19:
I think I got some new inside. Starting the app in windows linux bash shell using WSL2 overcomes the issue with the above stated METEOR_SETTINGS Error. Though, only if the app is started using the code in the README. If a env. file is used to pass `METEOR_SETTINGS=$(cat ./settings.json)' the same METEOR_SETTINGS Error as presented above appears. 
Unfortunately this solution does not fix the "demo-mode" error. Although I am using the settings.json file from the repository. As I understood correctly, this is a bug isn´t it? Or is there a way how I can fix this issue myself?
I´ll try to run the app on another workstation where a stand alone version of ubuntu is installed within the next days. Seems that only a fraction of user approach this demo-mode error. Thus I hope it´ll work in another machine.

adrianstrkr, 2021-05-19:
It Works! Sorry for the confusion. I just checked the version of my settings.json file again and realized that it was an older version than currently provided in the repository! With the up to date files the app works perfect on the Ubuntu bash shell on windows.
On first look, it looks like a perfect tool for our goals! So thank you again for your support and the development of such a great tool!

best wishes,
Adrian

closed 2021-05-19 by dmandrioli