Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #135: The problem of installing the sofware

MrCrazyCrab, 2021-05-20:
why i have to  Downloading meteor-tool@1.10.2.?? when i run the code of "meteor npm start", my system is windows10. i have tried many times and stocked here evertytime.

`C:\semantic-segmentation-editor-1.5.2>meteor npm install
up to date in 10.96s

4 packages are looking for funding
  run `npm fund` for details

C:\semantic-segmentation-editor-1.5.2>meteor npm start

> semantic-segmentation-editor@1.2.0 start C:\semantic-segmentation-editor-1.5.2
> meteor --settings settings.json --production

   Downloading meteor-tool@1.10.2...         [                            ] 0% 0.0s`

dmandrioli, 2021-05-21:
Meteor installation: https://www.meteor.com/developers/install

MrCrazyCrab, 2021-05-21:
i can install the meteor(2.2) but when i run the code meteor npm start, it will downloading the meteor-tool1.12. which i am stucking now

dmandrioli, 2021-05-21:
I'm not sure to understand what is going on for you but my experience of Meteor on Windows is not very good. You may want to deploy the tool on a linux machine or virtual machine.

MrCrazyCrab, 2021-05-21:
@dmandrioli  i have install it on ubuntu18.04 too, i got the same error. The version of meteor is mismatch with the software. The newest software(1.16) is developed on meteor 1.12,  but the meteor version i installed  through the code is 2.2.

closed 2021-05-28 by dmandrioli