Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #136: pcd file does not exist

soccken, 2021-05-31:
When I open the semantic-editor in a browser, only the default files exist.
How do I view files(pcd file) on my PC?

dmandrioli, 2021-05-31:
PCD and bitmap files are in "images-folder" as defined in the configuration file
See https://github.com/Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor#configuration-file-settingsjson

soccken, 2021-06-01:
Thank you, I was able to view the PCD file.
How can I personalize the label?
I want you to tell me how the label of setting.json works.

dmandrioli, 2021-06-01:
The README is not clear enough?
```
{
  "configuration": {
    "images-folder": "/mnt/images", // The root folder containing images and PCD files
    "internal-folder": "/mnt/pointcloud_data" // Segmentation data (only 3D) will be stored in this folder
  },
  // The different sets of classes available in the tool
  // For object classes, only the 'label' field is mandatory
  // The icon field can be set with an icon from the mdi-material-ui package
  "sets-of-classes": [
    {
      "name": "Cityscapes", "objects": [
      {"label": "VOID", "color": "#CFCFCF"},
      {"label": "Road", "color": "#804080", "icon": "Road"},
      {"label": "Sidewalk", "color": "#F423E8", "icon": "NaturePeople"},
      {"label": "Parking", "color": "#FAAAA0", "icon": "Parking"},
      {"label": "Rail Track", "color": "#E6968C", "icon": "Train"},
      {"label": "Person", "color": "#DC143C", "icon": "Walk"},
      {"label": "Rider", "color": "#FF0000", "icon": "Motorbike"},
      {"label": "Car", "color": "#0000E8", "icon": "Car"}
    },
    { ... }
  ]
}
```

soccken, 2021-06-01:
I want to build the label myself instead of the default label

dmandrioli, 2021-06-01:
Sure, you can define your own sets of classes by modifying the content of settings.json

soccken, 2021-06-01:
If you reduce the number of contents in the class set, the software will not start.
Please tell me how to change setting.json when I want to reduce the number of contents

dmandrioli, 2021-06-01:
Could you copy/paste here the content of your modified settings.json that prevents the tool to start?

closed 2021-06-02 by soccken