Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #139: How to use API

asiamanman, 2021-06-09:
Labeled PCD files are saved as .label, .objects.
I understand that this file is viewed using API Endpoints.
I would like to know the specific implementation method.

asiamanman, 2021-06-09:
I think I'll browse the file at localhost: 3000 / api / pcdtext / filename.
Is it possible to get a PCD file with only the labeled part?
Not all dots are labeled.

dmandrioli, 2021-06-09:
It's not possible to only get labeled points using the API. You will have to filter the result to only keep labeled points.

asiamanman, 2021-06-09:
Does that mean you can't just output what you label with PCD?

dmandrioli, 2021-06-10:
It just means that the output has the same shape than the input with 2 additional columns for labeling (object and class). It is then pretty easy to filter the output PCD file against a specific object or class value.

closed 2021-06-11 by dmandrioli