Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #144: Modifications to the application

ddudad, 2021-07-19:
Hey Damien,
I'd appreciate if you could tell me which scripts and what to change for the following modifications for the 3D editor

1.  I want to make a PCD file open button in the PCD Point Cloud Editor.
2.  now, only one class is allowed. I'd like to make several of the same car class. (for example car1 class, car2 class ..)

dmandrioli, 2021-07-20:
Hi @ddudad,

1. Currently PCD files are stored on the server, what you want to achieve would need too much work to be quickly described here.
2. Just edit your settings.json

ddudad, 2021-08-04:
thank you for your answer.

I have one more question

can i get distance(cm or m..) information using x,y,z coordinate of the points?

dmandrioli, 2021-08-04:
Hi @ddudad, this is not implemented.

ddudad, 2021-08-05:
I will try implement it.

I have a question

what is cloudData in SseData3d? 

is it different to x,y,z pos in .pcd?