Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #145: Point Cloud Editor unable to load large size .pcd files

madhavkarri, 2021-07-20:
Able to successfully load 19 MB file into the point cloud editor and visulaize the point cloud. 
But the editor is not able to load 1GB .pcd file.

dmandrioli, 2021-07-20:
Hello, there is no specific limitation on the file size. Performance mainly depends on your hardware and number of points in the .pcd.

Could you share a 1GB .pcd file to see what is the bottleneck?

madhavkarri, 2021-07-21:
Hello Damien,
Thank you for responding so quickly. Trying to get some approavals to share the 1Gb file.
In the mean time, is there an option to spread out the processes to multiple cpu cluster that will help load the 1 Gb file.
On my local machine I was able to load the 400 MB file, but the mouse operations are not quite smooth and there is a significant lag.

Jeehoon-K, 2021-09-15:
@madhavkarri 
Hi, I have the same problem with my pcd file. Did you fix your problem?

hop-soellingeraj, 2021-11-11:
I have the same problem.  My file has 20M points in it.  Can this tool handle that?  Does the tool use the GPU by default, because I am not seeing any activity on `nvidia-smi` as it loads the pcd file

madhavkarri, 2021-11-11:
From the tool author:
"It depends on your machine but I don't expect the tool to be able to load more than a few millions of points. I did some tests hacking the source code. On my Mac Mini of 2018, navigating the point cloud starts to be painful at about 5M of points"

madhavkarri, 2021-11-11:
This tool cannot handle pointcloud data beyond 2-3 million points.

madhavkarri, 2021-11-11:
It would not even load data for pointcloud data that contained roughly 16 million points

hop-soellingeraj, 2021-11-11:
Do you know what it would take to make it do larger point clouds, assuming my GPU is large enough?  I have a pretty large GPU on this workstation, and meshlab loads this file no problem.  It takes about 800MB of gpu memory.

hop-soellingeraj, 2021-11-11:
What I noticed was that there is no indication that the tool is using my GPU.  This GPU is a QuadroM4000, and it's the only one installed.  So it's not like the tool is loading onto another GPU.  I noticed inside the docker-compose there is no reservation for the gpu, which is typical of other gpu enabled containers I use.  I didn't know if the GPU support is supposed to be embedded in one of the dependencies and thus not visible in the repo.

madhavkarri, 2021-11-11:
I do not think it has to do with hardware limitations. Inherently it has to do with Hitachi tool limitations, rather than hardware.

INF800, 2022-02-24:
> What I noticed was that there is no indication that the tool is using my GPU. This GPU is a QuadroM4000, and it's the only one installed. So it's not like the tool is loading onto another GPU. I [...]

I opened new issue for the discussion

INF800, 2022-02-24:
@madhavkarri @hop-soellingeraj @dmandrioli . What was the conclusion of this issue.

dmandrioli, 2022-03-02:
@INF800 The tool relies on Three.js for point cloud display. Maybe some improvements are possible for this tool but the main limitation is Three.js.

closed 2022-03-02 by dmandrioli