Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #168: zooming smoothes image

gtlv97dot6, 2022-02-16:
Hi Damien,

 great tool you have here. I was trying it and noticed that when zooming all the way to the max on the picture I want to annotate, the editor is interpolating. I went to look in the code if there was something I could easily try and I saw that you already configured it this way from the very beginning. However, on Firefox and Chromium on linux, Firefox, Edge and, Chrome on Windows, there is smoothing despite your configuration.
 Not knowing the stack and not being a Node.js programmer, I figured it might be faster to ask you if you know what is going on rather than digging for hours.

 Thanks, for the software and for the help/fix if you can provide it.

 Cheers

gtlv97dot6, 2022-03-02:
Hello Damien, I managed to get some time to try some changes here and there. By updating paperjs to the most recent version I got back the expected behaviour in terms of smoothing, however it broke the placement of the image and I could not find a way to fix it. I tried all sorts of changes in imageLoaded and resizeCanvas but without success.

dmandrioli, 2022-03-02:
Thank you for your feedback. Actually I also tried some experiments without success...