Hitachi-Automotive-And-Industry-Lab/semantic-segmentation-editor #190: Unknown Error

cyy280113999, 2023-05-31:
**meteor npm install**
found 14 vulnerabilities (5 moderate, 8 high, 1 critical)
  run `npm audit fix` to fix them, or `npm audit` for details

**meteor npm audit fix**
+ three@0.152.2
+ underscore@1.13.6
+ body-parser@1.20.2
+ shelljs@0.8.5
added 25 packages from 14 contributors, removed 2 packages and updated 24 packages in 6.855s

21 packages are looking for funding
  run `npm fund` for details

fixed 12 of 14 vulnerabilities in 260 scanned packages
  2 vulnerabilities required manual review and could not be updated

**meteor npm audit**
                       === npm audit security report ===

┌──────────────────────────────────────────────────────────────────────────────┐
│                                Manual Review                                 │
│            Some vulnerabilities require your attention to resolve            │
│                                                                              │
│         Visit https://go.npm.me/audit-guide for additional guidance          │
└──────────────────────────────────────────────────────────────────────────────┘
┌───────────────┬──────────────────────────────────────────────────────────────┐
│ Moderate      │ Got allows a redirect to a UNIX socket                       │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Package       │ got                                                          │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Patched in    │ >=11.8.5                                                     │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Dependency of │ download                                                     │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Path          │ download > got                                               │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ More info     │ https://github.com/advisories/GHSA-pfrx-2q88-qq97            │
└───────────────┴──────────────────────────────────────────────────────────────┘
┌───────────────┬──────────────────────────────────────────────────────────────┐
│ High          │ http-cache-semantics vulnerable to Regular Expression Denial │
│               │ of Service                                                   │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Package       │ http-cache-semantics                                         │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Patched in    │ >=4.1.1                                                      │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Dependency of │ download                                                     │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Path          │ download > got > cacheable-request > http-cache-semantics    │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ More info     │ https://github.com/advisories/GHSA-rc47-6667-2j5j            │
└───────────────┴──────────────────────────────────────────────────────────────┘
found 2 vulnerabilities (1 moderate, 1 high) in 283 scanned packages
  2 vulnerabilities require manual review. See the full report for details.

**meteor npm start**
> semantic-segmentation-editor@1.6.0 start /home/cyy/tools/semantic-segmentation-editor
> meteor run --settings settings.json --exclude-archs "web.browser.legacy, web.cordova"

[[[[[ ~/tools/semantic-segmentation-editor ]]]]]

=> Started proxy.
/home/cyy/.meteor/packages/meteor-tool/.1.12.0.rmh4j4.m6y4n++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/dev_bundle/lib/node_modules/meteor-promise/promise_server.js:218
      throw error;
      ^

SyntaxError: Unexpected end of JSON input
    at JSON.parse (<anonymous>)
    at /home/cyy/.meteor/packages/meteor-tool/.1.12.0.rmh4j4.m6y4n++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/tools/fs/tools/fs/optimistic.ts:321:17
[... 36 lines cut ...]
    at /tools/isobuild/bundler.js:3222:24
    at Object.capture (/tools/utils/buildmessage.js:283:5)
    at bundle (/tools/isobuild/bundler.js:3169:31)
    at /tools/isobuild/bundler.js:3113:32
    at Slot.withValue (/home/cyy/.meteor/packages/meteor-tool/.1.12.0.rmh4j4.m6y4n++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/dev_bundle/lib/node_modules/@wry/context/lib/context.js:73:29)
    at Object.withCache (/home/cyy/.meteor/packages/meteor-tool/.1.12.0.rmh4j4.m6y4n++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/tools/fs/tools/fs/files.ts:1659:39)
    at Object.bundle (/tools/isobuild/bundler.js:3113:16)
    at /tools/runners/run-app.js:572:24
    at Function.run (/home/cyy/.meteor/packages/meteor-tool/.1.12.0.rmh4j4.m6y4n++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/tools/tool-env/tools/tool-env/profile.ts:289:14)
    at bundleApp (/tools/runners/run-app.js:571:34)
    at AppRunner._runOnce (/tools/runners/run-app.js:617:35)
    at AppRunner._fiber (/tools/runners/run-app.js:931:28)
    at /tools/runners/run-app.js:401:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! semantic-segmentation-editor@1.6.0 start: `meteor run --settings settings.json --exclude-archs "web.browser.legacy, web.cordova"`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the semantic-segmentation-editor@1.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/cyy/.npm/_logs/2023-05-31T07_46_36_604Z-debug.log

**/home/cyy/.npm/_logs/2023-05-31T07_46_36_604Z-debug.log**

lonelybuffer, 2023-09-06:
My ERRs are the same as yours. Have you solved it yet?

cyy280113999, 2023-09-06:
maybe the version of package.
i tried to reinstall and not update the meteor, it works

lonelybuffer, 2023-09-06:
> maybe the version of package. i tried to reinstall and not update the meteor, it works
Thank you replay to me. I installed the semantic-segmentation-editor-1.6.0 and meteor 1.11, but it does not work. Can you tell me which version you installed?

lonelybuffer, 2023-09-06:
Thank you again.
I think I found the real problem, just now. 
```
=> Started proxy.
/home/cyy/.meteor/packages/meteor-tool/.1.12.0.rmh4j4.m6y4n++os.linux.x86_64+web.browser+web.browser.legacy+web.cordova/mt-os.linux.x86_64/dev_bundle/lib/node_modules/meteor-promise/promise_server.js:218
throw error;
``` 
For solving the problem, I update my meteor to 2.5.3. I don't change anything else. Then it's work.
I refer to this link https://github.com/meteor/meteor/issues/11839#issuecomment-1007391404

cyy280113999, 2023-09-06:
oh, the latest meteor fix this.